dcs-bios/dcs-bios-arduino-library #4: Remove callbacks from ExportStreamListeners

Gadroc, 2015-01-24:
Currently some of the listeners have callbacks.  These callbacks are called during protocol parsing which is extremely timing sensitive at higher bus speeds.  We should move these out and any code that was executed there should be triggered at end of frame.

danvac, 2015-11-19:
Is there any guideline how to work with StringBuffers after this change?

jboecker, 2015-11-21:
I recommend using the release versions. I have not tested the code in the current master branch much, most of my development time has been spent on implementing RS-485 support. When I get that done, I will integrate it and release a new, tested version.

jboecker, 2017-05-04:
Current versions only copy the data to another buffer during parsing, callbacks are called from DcsBios::loop().

closed 2017-05-04 by jboecker