dcs-bios/dcs-bios-arduino-library #10: Crash on restart?

OlHall, 2015-02-16:
Hi,

I've noticed that the Arduino crashes under certain circumstances.
If the DCS simulation is running, and THEN I start the connect-serial-port.cmd script, the Arduino crashes, and sometimes restarts a bit later, sometimes it seems to hang forever.
If I start the script and THEN start DCS running (or unpause) it all works ok.
Any ideas what this might be? Is the parser blowing up on a "partial" read?

jboecker, 2015-02-16:
It shouldn't do that (for example, if you watch the DCS-BIOS intro video, I do exactly that without problems).

The parser and protocol are designed to handle the occasional error in the data stream -- the next update ~30 ms later will resynchronize it and any wrong data will be corrected within about 10 seconds by the "auto-sync" mechanism that occasionally re-sends values even if they did not change.

Check the rest of your code to see if you have a buffer overflow somewhere. Those can lead to all sorts of weird behavior (and cause crashes long after they happen).

OlHall, 2015-02-16:
This was using the standard example sketch with a single pot defined (so I could see the transmit light flickering away).  None of my own code at all.
I'm using an Arduino Mega 2560 R3 - if that makes any difference.

OlHall, 2015-02-20:
I'm not seeing this anymore, so I'll close it

closed 2015-02-20 by OlHall