dcs-bios/dcs-bios-arduino-library #9: Potentiometer "Jitter"

OlHall, 2015-02-16:
Would it be possible/correct to provide a mechanism to allow a pot object to be defined with a "jitter" band.  As it is, pots seem to generate a LOT of clutter on the output stream.

i.e. Add an optional parameter to the constructor for a "jitter band" value
In the PollingInput method, test the new value against the previous value and ONLY if it falls outside the jitter band
i.e. ( value < ( lastState_ - jitter )) || ( value > ( lastState_ + jitter ))
/* need to test for under/overflow when applying +/- jitter */

If true, send the update message, and set the new "lastState_".
In the current code, you always set the value of lastState_ even if it was the same as the previous read?

jboecker, 2015-02-16:
Yes, this definitely needs some work. Pull requests are welcome.

jboecker, 2017-05-04:
fixed in v0.2.10 release

closed 2017-05-04 by jboecker