dcs-bios/dcs-bios-arduino-library #13: OT: Newbie Question on Arduino IDE & Libraries

OlHall, 2015-02-17:
I'm new to Github and the Ardunio dev environment so apologies if this is a really stupid question:

I've forked the library to a working copy on my machine, but I noticed that the library that I imported into the Arduino IDE is structured very differently (monolithic .cpp/.h) - how do I build the library working copy so that I can then test the library with a sketch?

jboecker, 2015-02-17:
The current master branch here on GitHub already includes some restructuring work by Gadroc. The version you have is [v0.1.3](https://github.com/dcs-bios/dcs-bios-arduino-library/tree/v0.1.3), which still has the monolithic file structure.

closed 2015-02-17 by jboecker

OlHall, 2015-02-17:
I know this is closed, but...
I'm working on the master branch and it fails to compile - is that to be expected?

StringBuffer complains about incorrect number of arguments and Protocol.cpp has 'undefined reference to `onDcsBiosFrameSync()'

I'm just not sure if I have my environment incorrectly setup or the current master branch is broken?

jboecker, 2015-02-17:
Current master might be broken (it's been a while since I had time to test it).