dcs-bios/dcs-bios-arduino-library #18: v0.2 Roadmap

jboecker, 2015-03-21:
For v0.2 of the Arduino library, I want to have two major changes:

* Interrupt-based handling of all protocol handling (without the Arduino HardwareSerial library). This should eliminate the current problems we see with lots of outputs or outputs that take a long time to update (like displays). It is also a prerequisite for the next change:
* Support for half-duplex multidrop communication (over RS-485). We will define a protocol to have a bus master (the PC) poll the devices on the bus for input data in the time between sending out export data updates.

The  serial communication speed will be standardized as 250000 bps. That is fast enough to handle the data from DCS and can be generated precisely from a 16 MHz or 8 MHz system clock.

Here is what I expect v0.2 to include:
* Two modes of operation: Asynchronous full-duplex (the current one, where the device has a dedicated TX line to the PC) and half-duplex multidrop. Each mode will have its own template sketch. Because the interrupt-based serial port handling is handled differently on different microcontrollers, those template sketches will only support ATMega328-based Arduino boards (we can add support for Mega 2560-based boards later).
* A "legacy" template sketch for asynchronous full-duplex mode that relies on the Arduino HardwareSerial library. This will come with all of the problems of the current implementation, but it should run on just about any Arduino-compatible board (e.g. a Leonardo, Teensy or a Texas Instruments Tiva C launchpad).
* An expanded version of Jerker Dahlbloms COM Handler program that acts as the bus master on a half-duplex multidrop connection.

To implement these features, we will need the following changes (I will link these to separate issues later):
* Implement interrupt-based communication
* Define a protocol for half-duplex multidrop communication
* Change the sendDcsBiosMessage API so that sendDcsBiosMessage() is allowed to fail in case the transmit buffer is full (it will have a boolean return value to indicate success or failure). To ensure that every PollingInput instance gets a chance to transmit eventually, the PollingInput that is polled first should be changed whenever the transmit buffer becomes available again. The control classes (Switch2Pos etc.) will be modified to deal with a failing sendDcsBiosMessage() by trying to send the message again on the next call to pollInput().
* Sort the list of ExportStreamListeners by the lowest address the listener is interested in and add a "length" attribute to ExportStreamListeners to indicate the maximum address that it is interested in. This can be used to ensure that we will only iterate over that list once per incoming export data update.

jdahlblom, 2015-03-24:
I'm game for extending the COM Handler. Do you guys keep TS meetings 
when discussing more complicated stuff or just in writing?

jboecker, 2015-03-28:
The general design is a result of some troubleshooting I did with Gadroc shortly after DCS-BIOS was first released. Some of that happened over PMs on the ED boards.

Nowadays I use GitHub issues for planning.

OlHall, 2015-03-30:
Hi there,
What's the rationale behind using 485?  I'm just wondering why this would be preferable to making one Arduino 232 master over USB and then do inter-board comms using SPI?
I know 485 transceivers aren't expensive, but it adds complexity to the build, both at the PC (going 232 to 485) and for each board (requiring a transceiver), possible termination issues etc...

P.S. (we do a LOT of 485 in our work and when it works it's great.  When it doesn't it can be a PITA).

jboecker, 2015-03-30:
If you want the full story, check the original DCS-BIOS announcement thread. There were concerns about the reliability of communicating at 250000 bps using single-ended signaling. (Similar to your approach, I was initially recommending TTL UART for export data and I2C for sending input to DCS.)

I am not an electrical engineer and I don't have a simpit of my own to actually test this stuff. I still need to solve the problem of connecting several boards without using a USB port for each one (because that's the question everybody has after they run out of I/O pins on their first board).

RS-485 seems like an industry-proven standard with plenty of "safety margin" for a typical home cockpit. At 250000 bps it can handle a bus length of about 40 meters according to application notes from Maxim and TI. A simpit build will probably only have a bus length of under 5 meters.

I want to recommend something that is affordable, simple enough for others to build, and known to be able to scale up to the amount of boards and bus length that would be used in a complete simpit. I don't want to risk recommending a solution that works initially and then starts to fail intermittently in hard-to-debug ways once you add enough boards.

Using only built-in features of the Arduino board (the UART, SPI and I2C) is even cheaper and simpler, but more likely to break unpredictably. In some home-cockpit builds, there can be a lot of interference caused by PWM-controlled backlighting or possibly even a real instrument that is supplied 400 Hz AC power. If I were building a pit for myself, I might go with this approach, but I also have a logic analyzer and enough knowledge about the system to use it to find the cause of communication errors.

The Arduino library itself will only be concerned with supporting a protocol for half-duplex communication. The template sketch will still have the final say about how communication with DCS-BIOS is implemented, so by substituting a different template sketch you will still be able to use a dedicated serial connection per board, an Ethernet shield or whatever else you want to implement.

OlHall, 2015-03-30:
I know what you mean, and 485 will go for miles (well.. hundreds of feet at least) - I suspect way more than you'd need for a simpit.  It's definitely robust too.  We do control systems integration for expensive houses, so we come across 485 in lots of forms (everything from lighting control to heating systems to keypads) and people call up a LOT with problems about not getting any working comms between the processor (e.g. a PC) and the remote nodes.  The usual issues are polarity problems, which isn't helped by the fact than none of the transceiver mfrs stick to a common nomenclature.  A is *supposed* to be the -VE leg with B being the +VE, but many chip mfrs (maxim included I think!) have used A +VE, B -VE.  That can cause a load of worry.

There's often a maximum no of transceivers on a bus too - usually 32 but transceivers differ - that'll be on the specs, but something else to bear in mind.

Then there's the termination question.  Officially, you should use termination but most of the time you don't need to, and sometimes it's positively bad!

RS232 to 485 converters are pretty cheap from ebay, but they need power, and although some claim they can be powered from the RTS/CTS pins, they often can't and need an external 5-12v supply.

I'm not saying it's not a good way to go, and I have to admit I don't know a lot about SPI support on Arduino boards (we're more PIC focused) but we've done some board-to-board SPI stuff in a fairly noisey/high power environment and it worked perfectly - only anecdotal I know - so take it with a pinch of salt.

Don't get me wrong, I don't mind which you go - and looking forward to trying it out.  I've already maxed out my Arduino Mega ;-)

OlHall, 2015-03-30:
P.S. Where's the original DCS-BIOS thread?

jboecker, 2015-03-31:
The original announcement thread is [this 35-page monster](http://forums.eagle.ru/showthread.php?t=136570). RS-485 is first mentioned on page 4, in [this post from Gadroc](http://forums.eagle.ru/showthread.php?p=2280827#post2280827).

On page 12, [we start discussing problems with the current non-interrupt based implementation](http://forums.eagle.ru/showthread.php?p=2295795#post2295795).

[In this post,](http://forums.eagle.ru/showthread.php?p=2319734#post2319734) Gadroc estimates the number of Arduino boards that are required for a full cockpit to be about 60.

A few posts later, [Mike Powell also strongly recommends going with RS-485.](http://forums.eagle.ru/showthread.php?p=2319929#post2319929).

Your insight on working with RS-485 buses is very helpful! It looks like we can avoid most of these problems by using the same transceiver chip for all boards. I plan to recommend the MAX487. It has a 1/4 unit-load, so you can put 128 of those on the same bus, and it is "slew-rate limited", which according to the datasheet makes it especially robust in the face of interference or poor termination (at the cost of having a "slow" maximum transmission speed of 250000 bps).

OlHall, 2015-04-14:
Feature Request:  Is it possible to identify the first update/export when DCS starts so that we could have the various ExportListeners do an "auto-update".  At the moment, I have to flick each switch/turn a pot in order for it to update the sim.

jboecker, 2015-04-15:
I have opened a new issue to track your feature request (see #20).

closed 2017-05-04 by jboecker