dcs-bios/dcs-bios-arduino-library #24: Potentiometer not working

amyandpeteb, 2016-04-30:
I use the potentiometer in DCS BIOS but after adjusting it a few times, it crashes DCS BIOS.  Any suggestions?

jboecker, 2016-05-03:
Looks like some device IDs for the Huey have changed. That probably means a lot of other controls have stopped working as well.

amyandpeteb, 2016-05-04:
Thanks.  I just posted some info in another thread.  Here is my sketch (below).  The issue is with the ILS volume (it is the only potentiometer that I am using).

#define DCSBIOS_DEFAULT_SERIAL
#include "DcsBios.h"

/* Set Master Arm switch to Pin 0 and 1 */
DcsBios::Switch3Pos ahcpMasterArm("AHCP_MASTER_ARM", 0, 1);

/* Set GunPac Arm switch to Pin 2 and 3 */
DcsBios::Switch3Pos ahcpGunpac("AHCP_GUNPAC", 2, 3);

/* Set Laser Arm switch to Pin 4 and 5 */
DcsBios::Switch3Pos ahcpLaserArm("AHCP_LASER_ARM", 4, 5);

/* Set TGP switch to Pin 6 */
DcsBios::Switch2Pos ahcpTgp("AHCP_TGP", 6);

/* Set CICU switch to Pin 7 */
DcsBios::Switch2Pos ahcpCicu("AHCP_CICU", 7);

/* Set CICU switch to Pin 8 */
DcsBios::Switch2Pos ahcpJtrs("AHCP_JTRS", 8);

/* Set IFFCC switch to Pin 9 and 10 */
DcsBios::Switch3Pos ahcpIffcc("AHCP_IFFCC", 9, 10);

/* Set Autopilot mode switch to Pin 11 and 12 */
DcsBios::Switch3Pos lasteApMode("LASTE_AP_MODE", 11, 12);

/* Set Autopilot engage / disengage switch to Pin 13 */
DcsBios::Switch2Pos lasteApToggle("LASTE_AP_TOGGLE", 13);

//this is a test...I'm having problems with the potentiometer and I want to check it out with the ILS volume A0
DcsBios::Potentiometer ilsVol("ILS_VOL", A0);

void setup()
{
  DcsBios::setup();

}

void loop()
{
  DcsBios::loop();

}

jboecker, 2016-05-07:
Any particular reason for using DCSBIOS_DEFAULT_SERIAL instead of DCSBIOS_IRQ_SERIAL?

The underlying issue here is that I still haven't found the time to write a proper Potentiometer class that does some averaging, threshold, and rate-limiting of updates, but at least with DCSBIOS_IRQ_SERIAL it has no chance of causing things to lock up completely.

The issue with DEFAULT_SERIAL is that while a message is being sent to DCS-BIOS, incoming data is not being processed, so the receive buffer will overflow if messages are being sent all the time because the pot is a bit noisy.

amyandpeteb, 2016-05-07:
Now I understand.  I'm using the Leonardo board, which requires me to use the IRQ_SERIAL otherwise it will not compile.  So I guess my option is to wait for a fix or go with the UNO board?  I used the LEONARDO as i had thought I would use some additional features on it, like tying buttons to keyboard commands.

closed 2016-05-08 by amyandpeteb