dcs-bios/dcs-bios-arduino-library #28: Library does not compile for the new Arduino Nano Every or any architecture that is not AVR

SaabFAN, 2019-08-18:
Will you add support for the new Nano Every?

I just tried compiling the Master Caution-Example in the Arduino IDE and got loads of errors.
The Sloeber IDE - Eclipse modified to behave more like the Arduino IDE - doesn't even find the library when I try to compile the example for the Nano Every.

Here are the compiler-errors I get when trying to compile in Arduino IDE:

[CompilerErrors_ArduinoIDE_NanoEvery.txt](https://github.com/dcs-bios/dcs-bios-arduino-library/files/3513607/CompilerErrors_ArduinoIDE_NanoEvery.txt)

SaabFAN, 2019-08-18:
Errors remain with and without Register Emulation for Atmega 328 turned on.

SaabFAN, 2019-08-18:
Problem is related to the Interrupt-Based Serial communication. 
If "DCSBIOS_DEFAULT_SERIAL" is defined, the error vanishes and the example compiles in Arduino IDE. 

Compilation still fails in Sloeber, despite "DCSBIOS_DEFAULT_SERIAL" being defined <- Issue in Sloeber.

Day0Dreamer, 2019-09-14:
Any fix to this found?

SaabFAN, 2019-09-14:
The IRQ-Based methods are too hardware-specific. 
The fix would be to rename all the relevant registers of the Serial Port inside the Microcontroller. 
For the Leonardo it is probably also necessary to rewrite some additional code because the hardware is completely different. 
Same goes for the STM32 "Bluepill" or any other Micro that does not have the original AVR-Core. 

Regarding Sloeber and why it is still failing during compilation, I have no idea.

SaabFAN, 2019-09-14:
Quick addition for people using Sloeber: Creating the Project as a "Default CPP"-Project, not as an ".Ino"-Project solves the problem with compiling when using "DCSBIOS_DEFAULT_SERIAL".

jboecker, 2019-11-05:
I have no plans to add support for any other Arduino boards. Porting the code to a new board means I have to support that board for all eternity, and that cost is usually not justified because that new board does not make anything possible that wouldn't have been possible before by just throwing another Nano into the mix.

The ATMega328 controller is an obvious choice because there are a lot of cheap chinese clones available. Anyone can get one for cheap, no matter where they are in the world.

The ATMega2560 was low-hanging fruit and has a significantly larger amount of GPIO pins.

The only other controller I might consider, after completing a bunch of other features first, is the STM32C8T6 ("blue pill" boards), as it costs about the same as an Arduino Nano but has enough processing power for more CPU intensive applications, such as driving larger graphical displays, and most of its I/O pins are 5V-tolerant.

closed 2019-11-05 by jboecker