dcs-bios/dcs-bios-arduino-library #32: Ka-50 Any way to get PVI-800 lines as strings inclusive of apostrophes?

virtual812, 2020-10-19:
Feeding PVI-800 data to a display that is addressed as a serial device and takes input as a string.
DCS BIOS -> (Serial0)Arduino Mega2560(Serial1) -> display
The standard outputs from DCS BIOS do not include the apostrophes in the 'Display top line text' or 'Display bottom line text' strings.
Any way this can be remedied?

virtual812, 2021-06-02:
Anyone still alive in here?
Would love to simplify the way i feed the PVI-800 data to a display.

talbotmcinnis, 2021-06-02:
I’m monitoring this ad well but any development would be done on the talbotmcinnis fork.  

To date there are no aircraft specific changes in the dcs-bios arduino library and I think its good to keep it that way if possible.  Also there is not string processing for quote removal, so I suspect this may require changes on the dcs-bios side, if at all.  I don’t know the ka-50 but I’d bet you get the raw string that dcs provides.  I’m speculating a lot here.  Can you use the controls reference to give examples of the desired changes?

virtual812, 2021-06-02:
As it is currently DCS BIOS treats the numbers, the apostrophes and the plus or minus symbol as separate pieces of data.
This is fine if for example you're running 7 segment LED displays, and have separate LEDs for the apostrophes and a separate display for the plus or minus.
But in our case we are driving an LCD with a string of text, and at this point we have to process the text depending on weather or not there are apostrophes and +/- sign in the arduino.
Would be great to simply have a string from DCS that includes the punctuation so we can display it as is.

PS - I edited my reply on your github re the arduino library STM possibility and tried to link the errors i was getting in a more readable format, but for some reason github wouldn't show the edit, even though when i clicked edit again it was present in the modified form. It also wouldn't allow an additional post there to explain the issue, will follow up there later.