dcs-bios/dcs-bios-arduino-library #34: Steps Per Detent not working as described in documentation

ColdFrontWI, 2021-05-09:
I'm trying to use some encoders that are two steps per detent vs. the typical four.

In the documentation, it indicates that you can add <2> after the declaration to work with a two-step encoder, e.g. change
~~~
DcsBios::RotaryEncoder tacan("TACAN_1", "DEC", "INC", 4, 5);
~~~
to
~~~
DcsBios::RotaryEncoder<2> tacan("TACAN_1", "DEC", "INC", 4, 5);
~~~
but this throws an "is not a template" error:
~~~
Simple_DCS_Bios_Test:17:1: error: 'DcsBios::RotaryEncoder' is not a template
 DcsBios::RotaryEncoder<2> tacan("TACAN_1", "DEC", "INC", 4, 5);
 ^~~~~~~
exit status 1
'DcsBios::RotaryEncoder' is not a template'
~~~

I have had success editing encoders.h by changing
~~~
RotaryEncoder(const char* msg, const char* decArg, const char* incArg, char pinA, char pinB, StepsPerDetent stepsPerDetent = FOUR_STEPS_PER_DETENT) {
~~~
to
~~~
RotaryEncoder(const char* msg, const char* decArg, const char* incArg, char pinA, char pinB, StepsPerDetent stepsPerDetent = TWO_STEPS_PER_DETENT) {
~~~
but this makes the change universal, and I have many 4-step encoders in my design.

Is there a way to specify only certain controls as 2-step encoders?

Thanks,
Eric

talbotmcinnis, 2021-05-09:
Part 1: DcsBios::RotaryEncoder is already a default instantiation of the template.  To make your own, do one of the following:

typedef DcsBios::RotaryEncoderT<POLL_EVERY_TIME, DcsBios::TWO_STEPS_PER_DETENT> MyTwoStepRotary;
and then your control declaration is
MyTwoStepRotary uhf10mhzSel("UHF_10MHZ_SEL", "DEC", "INC", 7, 5);

Or you can do it all on one line if you prefer:
DcsBios::RotaryEncoderT<POLL_EVERY_TIME, DcsBios::TWO_STEPS_PER_DETENT> uhf10mhzSel("UHF_10MHZ_SEL", "DEC", "INC", 7, 5);

Part 2 is that is not clear enough in the documentation, which I'll clean up.  FYI the active and supported version has been forked here: https://github.com/talbotmcinnis/dcs-bios-arduino-library

talbotmcinnis, 2021-05-09:
Can you give me a link to the documentation you are referring to please?

ColdFrontWI, 2021-05-09:
Thanks for the info!

I found a messier hack to do it by adding a "RotaryEncoder2" class that specified two steps per detent.  Will look at the approach you describe today.  I also missed the fork, so thanks for pointing that out as I've been using an older version.

As for the documentation, found it referenced in a couple of places.  Section 6.6 on p29 of the documentation at [https://dcs-bios.readthedocs.io/_/downloads/en/latest/pdf/](https://dcs-bios.readthedocs.io/_/downloads/en/latest/pdf/).

It's also referenced in the "RotaryEncoderFixedStep" section at [https://dcs-bios.readthedocs.io/en/latest/code-snippets.html](https://dcs-bios.readthedocs.io/en/latest/code-snippets.html).

Thanks again,
Eric

talbotmcinnis, 2021-05-09:
Ok thanks for that link.  That is the documentation for the unsupported "hub" version so I cannot update that. I really should create something similar for the fork I support. One day :)

Glad you got something to work.  The other fork is not majorly different, so stick with what works for now.