dcs-bios/dcs-bios-arduino-library #35: F16 DED

Wombat82, 2021-08-24:
Hello, my project is the F16 DED. I tried my way around programming. I also get the display, but as soon as I use it, the output throws everything into confusion. Is there a chance that you will help to get the Code right.

talbotmcinnis, 2021-08-24:
I will do my best.  Can you share your sketch?

Wombat82, 2021-08-25:
My sketch."#define DCSBIOS_DEFAULT_SERIAL#include "DcsBios.h"#include <Arduino.h>#include <Wire.h>#include <U8g2lib.h>#include <U8x8lib.h>U8G2_SSD1309_128X64_NONAME0_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 13);void onDedLine1Change(char* newValue) {    /* your code here */    u8g2.setFont(u8g2_font_5x7_tf);    u8g2.drawStr(0,10,newValue);    u8g2.sendBuffer();        }DcsBios::StringBuffer<25> dedLine1Buffer(0x44fc, onDedLine1Change);void onDedLine2Change(char* newValue) {    /* your code here */    u8g2.setFont(u8g2_font_5x7_tf);    u8g2.drawStr(0,20,newValue);    u8g2.sendBuffer();}DcsBios::StringBuffer<25> dedLine2Buffer(0x4516, onDedLine2Change);void onDedLine3Change(char* newValue) {    /* your code here */    u8g2.setFont(u8g2_font_5x7_tf);    u8g2.drawStr(0,30,newValue);    u8g2.sendBuffer();}DcsBios::StringBuffer<25> dedLine3Buffer(0x4530, onDedLine3Change);void onDedLine4Change(char* newValue) {    /* your code here */    u8g2.setFont(u8g2_font_5x7_tf);    u8g2.drawStr(0,40,newValue);    u8g2.sendBuffer();}DcsBios::StringBuffer<25> dedLine4Buffer(0x454a, onDedLine4Change);void onDedLine5Change(char* newValue) {    /* your code here */    u8g2.setFont(u8g2_font_5x7_tf);    u8g2.drawStr(0,50,newValue);    u8g2.sendBuffer();}DcsBios::StringBuffer<25> dedLine5Buffer(0x4564, onDedLine5Change);void setup(void) {    u8g2.begin();   // Start OLED    u8g2.clear.Display();    DcsBios::setup();}void loop(void) {   DcsBios::loop();  }"

talbotmcinnis, 2021-09-05:
Hey. So looking at this the DCS bios usage looks right to me.  I suggest the following:

Remove the SendBuffer calls.  According to the documentation for your display controller, those are to be used with NextPage() for buffered display, which you are not using.  https://github.com/olikraus/u8g2/wiki/u8g2reference#drawstr

Simplify for testing.  Start with all code in setup, issue the clear, then try drawStr with known strings like "L2HERE".  Once that displays properly, move each line into the DCS callbacks but leave in the static strings.  If that works, Finally use the dcs newValue strings.

Let me know how that goes.

Wombat82, 2021-09-06:
Hey, i tried your solution but no results. The first sketch shows me the results in the pictures. The last picture is when i push Comm1 button.

talbotmcinnis, 2021-09-06:
I don't see the sketches or pictures.  Maybe easier to discuss on discord?

Puma [476th 76th vFS]#7227