dcs-bios/dcs-bios-arduino-library #37: Requesting adding compatability to the Seeedunio XIAO (arduino compatible)

mg102279, 2021-12-29:
Can you add the Seeeduino XIAO to be compatible to use with DCS-BIOS? It would help a lot. It is very small but would help a lot in the long run. Others may not know about this heavy processing device in a small package.
https://www.seeedstudio.com/Seeeduino-XIAO-Arduino-Microcontroller-SAMD21-Cortex-M0+-p-4426.html

talbotmcinnis, 2021-12-31:
If its actually Arduino compatible I’d expect it to work out of the box (non-irq mode).  Are you getting compilation errors?  If so please share the specifics.

mg102279, 2022-01-06:
It works perfectly fine in default_serial. Would like IRQ_serial with the XIAO. Beggars can't be choosers! lol.

Got the F-16 DED working, on it.  Now I have to figure out how to get the right font installed that will show the degrees ° symbol without a space after the LAT and LON, inverted text and highlight when entering data in DED. Coming together though.  Thanks for responding.