dcs-bios/dcs-bios-arduino-library #38: Encoder is skipping

wmundstock, 2021-12-30:
Hi All!

I'm using very simple code for 2 encoders for the F5E. The code is ultra simple and works very well, except that every now and then the encoder will skip. It does not seem to be related to the encoder itself as it happens with both of them.

On a quick research I found that the skips can be "minimized" with the way the encoders are implemented, but I'm not expert in such.

Any help is appreciated.

/* paste code snippets from the reference documentation here */
DcsBios::RotaryEncoder hsiCrsKnob("HSI_CRS_KNOB", "-3200", "+3200", PIN_HSI_CRS_ENCODER_DN, PIN_HSI_CRS_ENCODER_UP);
DcsBios::RotaryEncoder hsiHdgKnob("HSI_HDG_KNOB", "-3200", "+3200", PIN_HSI_HDG_ENCODER_DN, PIN_HSI_HDG_ENCODER_UP);

void setup() {
  DcsBios::setup();
  
}

void loop() {
  DcsBios::loop();
}

talbotmcinnis, 2021-12-30:
Are you using optical or mechanical encoders?  I’ve just started the conversion to optical after giving up on achieving reliable mechanical encoders on an arduino.

Its easy enough to add some debug statements to determine if the Arduino is even detecting the missed turns but I predict you will find it is not in which case there is little we can do from software.  How many does it miss?  1 in hundreds would suggest a noise or timing issue.  1 in 4 would suggest a logic or wiring issue.

wmundstock, 2022-01-03:
I thoroughly cleaned the encoder with alcohol and that sorted out most of it. Seems like there were some grease in the contacts causing extra skips. 

It still skip every now and then but it is no longer that noticeable. I would say one in dozen clicks. 
In addition it skips when I switch direction (up down).